eprive/Z80-MBC3 #4: Z80-MBC3 to assemble

hosamki, 2023-05-24:
I want to buy a Z80-MBC3 to assemble. What should I do?

hosamki, 2023-05-24:
I live in Seoul, South Korea.

eprive, 2023-05-24:
There is a webshop that is dedicated to the Z80-MBC3. You will find it here : https://eighties.company.site/

And you can choose the kit to be assembled, or a ready made and tested board.

You can use the email address ( eprive@gmx.com ) as mentioned under my account for future enquiries.

eprive

hosamki, 2023-05-30:
I tried to order on the site you told me, but it cannot be delivered to South Korea. 
Why?

hosamki, 2023-05-30:
There is not South Korea (or Republic of Korea) in Country of Shipping & Delivery.

hosamki, 2023-05-30:
I sent you a email.

hosamki, 2023-05-31:
I checked that "Republic of Korea" was added to the list of countries. 
So I just ordered and paid for two Z80-MBC3. 
I look forward to receiving it as soon as possible.

closed 2023-08-31 by eprive