eprive/Z80-MBC3 #9: Support

mciucci65, 2024-08-16:
Can you explain me better how to use external i2c bus and what is the meaning of JP1 connected to INT signal of i2c GPIO controller and share some code example ?.

Can you ask Carlo Consoli to share the code for LCD display? 

Is it possible to share the sketch? I want to implement a status display controlled by i2c bus.

eprive, 2024-08-16:
The way to use the I2C bus with the Z80-MBC3 is described in the application note : AN-GPIOandI2C.pdf

The on board PCF8574 is fixed at adress 0x20. External expansion with additional PCF8574 or MCP23017 should be set at other unique addresses. Standard LCD modules are available which have their own PCF8574 on a piggyback addon.
JP1 is unconnected and not used. It would be possible to have external interrupts when needed for customised applications.

I understand that Carlo Consoli has shared his code and I can ask him to post it on the Facebook group page or email it to you. This applies to all sample code created by Z80-MBC3 users.

The Z80-MBC3 is not an Arduino compatible board but one for Z80 applications, and DIY assembly of a single board computer. The ATmega4809 IOS controller currently does not implement a "sketch" as was the case with the Z80-MBC2.

In answer to you last note : If your type of application is a simple status display, or one with a GPIO expansion, then your are advised to use a Arduino module, or cloned Arduino board.  With these you can implement the desired "sketch" and use the Arduino libraries.

mciucci65, 2024-08-17:
if I understand correctly you advise me to use an arduino as an I2C slave which in turn controls the 1602 display connected to the same i2c bus.
but which C library do I have to use on the board to interface and send the command to the I2C bus? Arduini use Wire.h
I simulate with Arduino as master and another Arduino as slave:

MASTER:
#include <Wire.h>           // Include the Wire library for I2C communication

const int SLAVE_ADDRESS = 0x08;  // Define the I2C address of the slave
String messageToSend = "Hello, Arduino!";  // Message to send to the slave

void setup() {
  Wire.begin();  // Initialize the I2C as master
  Serial.begin(9600); // Initialize serial communication for debugging
}

void loop() {
  Wire.beginTransmission(SLAVE_ADDRESS); // Start communication with the slave
  Wire.write(messageToSend.c_str());     // Send the message
  Wire.endTransmission();                // End the transmission

  Serial.println("Message sent to slave: " + messageToSend);
  
  delay(1000);  // Wait for a second before sending the next message
}
SLAVE:
#include <Wire.h>                // Include the Wire library for I2C communication
#include <LiquidCrystal_I2C.h>   // Include the library for the I2C LCD

const int SLAVE_ADDRESS = 0x08;  // Define the I2C address for the slave

LiquidCrystal_I2C lcd(0x27, 16, 2); // Set the LCD I2C address (0x27 is common) and size

String receivedMessage = "";      // Variable to store received message

void setup() {
  Wire.begin(SLAVE_ADDRESS);      // Initialize the I2C as slave
  Wire.onReceive(receiveEvent);   // Register the receive event function
  
  lcd.init();                     // Initialize the LCD
  lcd.backlight();                // Turn on the backlight
  
  lcd.clear();                    // Clear the LCD screen
  lcd.setCursor(0, 0);            // Set cursor to first row, first column
  lcd.print("Waiting for msg");   // Display initial message
}

void loop() {
  if (receivedMessage.length() > 0) {
    lcd.clear();                  // Clear the previous message
    lcd.setCursor(0, 0);          // Set cursor to the first row
    lcd.print("Received:");       // Print a static message
    lcd.setCursor(0, 1);          // Move cursor to the second row
    lcd.print(receivedMessage);   // Print the received message

    receivedMessage = "";         // Clear the message after displaying
  }
}

// Function to be called when data is received from the master
void receiveEvent(int bytesReceived) {
  receivedMessage = "";  // Clear the previous message

  while (Wire.available()) {
    char c = Wire.read();   // Read each byte and append to the message
    receivedMessage += c;
  }
}

closed 2024-08-17 by eprive