eprive/Z80-MBC3 #10: IOS works, but no program/os works

gmu-code, 2025-02-08:
After i boot the system it displays on Putty:

```
Z80-MBC3. IOS (ATmega4809) - release S40427

IOS: GPIO with PCF8574 or MCP23017 is enabled
IOS: CP/M Autoexec is ON
IOS: Select boot mode or system parameters:
 0: No change (2)
 1: Autoboot
 2: Basic
 3: Forth
 4: Load OS from Disk Set 0 CP/M 2.2
 5: iHex Loader
 6: Change Z80 clock speed (-> 10 MHz)
 7: Toggle CP/M Autoexec (-> OFF)
 8: Change Disk Set 0 CP/M 2.2
 9: Edit RTC date + time
Enter your choice >

```
The red LED is on, the blue is blinking.
If i select one of the choices 2, 3, 4 or 5 as sample here "**3 Forth**" it displays:

```
IOS: Loading boot program (FORTH13.BIN)...
IOS: Z80 clock speed set to 5 MHz
```
All LEDs are off. 
There is no response over Putty.

On "**2 Basic**"

```
IOS: Loading boot program (BASIC47.BIN)...
IOS: Z80 clock speed set to 5 MHz

```
The red LEDs is on. 
There is no response over Putty.

Are you any idea - is it a typical error who can be resolved?

Thanks.

pa3esb, 2025-02-08:
I once had a Z80 that was very dirty. After months of non use it didn’t work any more. Cleaning with alcohol helped to solve the problem.

gmu-code, 2025-02-08:
@eprive 
Thanks for this information.
I will check it.

@pa3esb 
Thanks for this hint - but my Z80 is not dirty :-) 
It's look like new.

pa3esb, 2025-02-08:
[image]
[image]

This is how dirty my Z80 was.  Clearly a chip that had been in storage for a long time. And as it became a bit wet it shorted the signals to ground. Cmos has very high impedance.