eprive/Z80-MBC3 #11: Booting Fuzix from disk set 5

twi42, 2025-04-20:
Z80-MBC3 arrived and was easy to assemble as promised. Everything else worked as documented, but I wanted to try Fuzix using the more recent sdcard image (SD-S220718-R290823-v2) for MBC2.

The image works like the preinstalled sdcard (s220718-r240620_sd_v1) but the IOS still only shows 4 disk sets and no Fuzix visible, even though the card has DS5N01.DSK and DS5NAM.DAT accordingly.

Booting Fuzix via autoboot results immediate crash, apparently due to missing disk mappings. 

It's likely something simple, but I'd appreciate a hint.

eprive, 2025-04-20:
This is not unexpected behavior, since Fuzix was released for the Z80-MBC2. Fuzix requires modifications to the IOS. Such a modification was not implemented in the IOS for the Z80-MBC3, which prevents Fuzix from booting. Although such a modification may be possible, I see no practical advantage or useful application in relation to the effort required to realise it.
As far as I know, Fuzix is ​​a very limited emulation of a Linux like console terminal without examples.

Any practical application could be made more efficient and simpler with the available languages ​​and disksets.

For this reason, the IOS for Fuzix was not implemented for the MBC3. If someone can provide arguments for its usefulness, I will reconsider implementing such a feature. It is probably just as useful as Collapse OS.

twi42, 2025-04-20:
Thank you for the comprehensive explanation - after looking into Fuzix bit deeper I came to the same conclusion. Also just to understand, adding any additional entries to the diskset menu requires support from IOS? I'm current checking if I can run a Mobitex terminal ROM from 1992 (32K) and see the original debug output.

eprive, 2025-04-21:
You can substitute one of the available disksets with another set. The naming is retrieved from the SD file.

closed 2025-04-21 by eprive